rapp-project/rapp-api #7: Face detection service enhancement

etsardou, 2015-12-14:
The face_detection service should have one more parameter: A bool called "fast". If true it provides fast face detection suitable for tracking.

- [ ] C++
- [ ] JS
- [x] Python

This feature exists in the devel branch of rapp-platform

alexge233, 2015-12-16:
Overloaded or different class?

etsardou, 2015-12-16:
Just the same service with one extra input parameter.

alexge233, 2015-12-20:
@etsardou Manos can you please provide the actual JSON params? [The Services Wiki](https://github.com/rapp-project/rapp-platform/wiki/RAPP-HOP-services) appears to have the old version.

klpanagi, 2015-12-21:
@alexge233  we updated the Wiki with the new face_detection, **fast** parameter entry ([wiki](https://github.com/rapp-project/rapp-platform/wiki/RAPP-HOP-services)).

```js
face_detection( { file_uri: '' , fast: false } )
```

- Declaration name: fast.
- Type: Boolean

Sorry for any inconvenience caused.

lpenteri, 2015-12-21:
I added the new feature to the face detector of the JS API but was unable to deploy the devel branch of the rapp-platform, where you said the enhanced web service lies, in order to test it. Is that expected (the devel branch is broken?) or it should be able to get installed and deployed and I missed something during the installation? What I did was to run the platform installation script rapp-platform/rapp_scripts/setup/6_rapp_platform_setup.sh giving "devel" as argument.

etsardou, 2015-12-21:
The devel branch includes the changes as you can see [here](https://github.com/rapp-project/rapp-platform/blob/devel/rapp_face_detection/src/face_detection.cpp#L48) and it is not broken as evident [here](https://travis-ci.org/rapp-project/rapp-platform/branches). Try to run the unit/functional and integration tests to make sure everything runs smoothly. In them you should spot a test regarding the face detection ```fast``` feature (like [here](https://travis-ci.org/rapp-project/rapp-platform/builds/97410414#L1517) or [here](https://travis-ci.org/rapp-project/rapp-platform/builds/97410414#L2888) ).

Also @klpanagi and all, please do not make changes in the wiki pages or in the master branches as this will cause confusion. With this change in the wiki someone could search in the master branch to find the ```fast``` parameter and find nothing. I will prepare a mail later to discuss the deployment / development course we must follow, in order for API and Platform to be synchronized.

commit a6789841 (2016-01-29): merged devel with ortelio branch and added the fast feature in face recognition JS API service. Closes #7
commit f74f68c1 (2016-02-24): merged devel with ortelio branch and added the fast feature in face recognition JS API service. Closes #7
commit a7496778 (2016-05-20): merged devel with ortelio branch and added the fast feature in face recognition JS API service. Closes #7
closed 2016-01-29 by alexge233