rapp-project/rapp-api #8: Token-based authentication

alexge233, 2015-12-20:
@klpanagi 
Authentication (in C++) is hardcoded into qr_detection, AFAIK.
If we need token-based authentication, it must be implemented across the API/Lanugages.

This has nothing to do with SSL/TLS https sockets which should be implemented in the next version.

klpanagi, 2015-12-21:
We previously discussed on enhancing security, by using token-based authentication.

The approach is for the per-robot-platform Core Agent to hold the Platform Authentication tokens. This approach requires the RAPP-API to request authentication tokens from the core-agent and use em either on initialization or on each Platform Service Call.

@etsardou should we consider this under an upcoming milestone?

alexge233, 2016-01-28:
I will freeze this for now, but I think we should use user tokens for identifying users on the platform.

closed 2016-01-28 by alexge233