rapp-project/rapp-api #12: Text to speech API support

etsardou, 2015-12-21:
- [ ] C++
- [ ] JS
- [x] Python

[Info](https://github.com/rapp-project/rapp-platform/wiki/RAPP-Text-to-speech-using-Espeak-&-Mbrola#input--output)

alexge233, 2015-12-30:
@etsardou what's this? documentation or wiki?

etsardou, 2015-12-30:
This refers to the text-to-speech node. Info [here](https://github.com/rapp-project/rapp-platform/wiki/RAPP-Text-to-speech-using-Espeak-&-Mbrola) (or click the "info" above)

alexge233, 2016-01-29:
Please see [issue #233 on rapp-platform ](https://github.com/rapp-project/rapp-platform/issues/233) regarding this. We can't add base64 for raw audio data, please change to multiform/part as per the RFC standard.

klpanagi, 2016-03-22:
Status update on JS API?
The text_to_speech RappCloud module exists [here](https://github.com/rapp-project/rapp-api/blob/ortelio/js/includes/cloud/text_to_speech.js).

lpenteri, 2016-03-23:
Yes it is implemented where you pointed, I just forgot to check the checkbox.

closed 2016-03-23 by lpenteri