rapp-project/rapp-api #16: Python tests migration

alexge233, 2016-03-13:
@klpanagi could you please migrate the Python tests under the API? AFAIK they are currently under the platform.

etsardou, 2016-03-13:
Actually that is not that easy since all the setup scripts and CI scripts have to change. I propose to revisit this in the next release (0.7).

alexge233, 2016-03-13:
If this is an API test (unit test) it should be in there, if its a platform test, then let's move them all to the platform?

etsardou, 2016-03-13:
Didn't get you. What you mean by "python tests"? If you mean [these](https://github.com/rapp-project/rapp-platform/tree/master/rapp_testing_tools), they are integration tests, as they test the API, the web services and the ROS background nodes.

alexge233, 2016-03-13:
I thought we had python unit tests, never mind them, nothing to move to the api for python.

closed 2016-03-13 by alexge233