rapp-project/rapp-api #19: [C++]: Use CMAKE_INSTALL_PREFIX var to assign installation paths

klpanagi, 2016-03-28:
Consider assigning installation paths using CMAKE_INSTALL_PREFIX argument to cmake.

maciek-slon, 2016-03-28:
I've created proper fixes and additions in CMake files, and I'm ready to apply them. But, as there are two active branches with C++ changes in the rapp-api (ortelio and devel), I don't know which one to use as a base for C++ code changes. I've asked Alex twice (19/03 and 22/03) which one to use, but got no answer, thus the patch is still waiting.

alexge233, 2016-03-28:
@klpanagi I will pass a CLI parameter to be used as the install directory for the libraries, and another for the headers, which will default to `/usr/local/` if nothing is passed.

@maciek-slon Please create a new branch `wut` and apply your changes there. We will merge once we've finished working on ortelio.

maciek-slon, 2016-03-28:
Should we base on ortelio or devel? Which one is "better" C++ base?

alexge233, 2016-03-28:
I don't know who owns and works on devel, ortelio will have our latest code but is not stable and will be changing significantly very soon. I would suggest you either clone ortelio and expect to have changes after, or wait until we do the move from HTTP to JSON.

klpanagi, 2016-03-28:
@maciek-slon  I would suggest the following:

- Create a new branch having **api_stable** as parent
- Rebase to Ortelio to get the functionalities provided there.
- Continue developing under the newly created branch, lets say cpp/<feature>

maciek-slon, 2016-03-30:
Unfortunatelly, rebasing to ortelio creates A LOT of conflicts in cpp/hpp files.

alexge233, 2016-04-28:
Added feature and closing.
Can somebody please test the `-DOPEN_NAO=On` flag please?

closed 2016-04-28 by alexge233

maciek-slon, 2016-05-12:
There is a typo in CMakeLists for OPEN_NAO: Line 180 should be 
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=gnu++11 -Wall -fPIC -static-libstdc++ -m32")
instead of 
set(CMAKE_CXX_FLAGS ${CMAKE_CXX_FLAGS} "-std=gnu++11 -Wall -fPIC -static-libstdc++ -m32")