rapp-project/rapp-api #25: Empty response in Current Weather - Python API

etsardou, 2016-05-25:
Request in ROS node:
```
city: Athens
weather_reporter: ''
metric: 0
```

Response in ROS node:
```
{'wind_speed': '0.32', 'temperature': '59.53', 'wind_direction': '', 'visibility': '9.05', 'humidity': '0.88', 'pressure': '1022.65', 'date': '1464173287', 'weather_description': 'Clear', 'wind_temperature': ''}
```

Response of Python API call:
```
{'wind_speed': '', 'temperature': '', 'wind_direction': '', 'visibility': '', 'humidity': '', 'pressure': '', 'error': u'', 'date': '', 'weather_description': '', 'wind_temperature': ''}
```

etsardou, 2016-05-25:
Branch ```news_weather_fixes``` in platform

klpanagi, 2016-05-26:
RAPP Platform Web Service issue. Resolved.
Closing this issue

closed 2016-05-26 by klpanagi