rapp-project/rapp-api #26: Text to speech should take as input the path to store the file

etsardou, 2016-05-25:
https://github.com/rapp-project/rapp-api/blob/python/python/RappCloud/CloudMsgs/README.md#text-to-speech

The example is unclear. First make the call and then add the path in the request?

klpanagi, 2016-05-25:
Thank you for mentioning this @etsardou . This is actually a **not-working example**

Audio data are returned as part of the Response Object.
The `TextToSpeech.Response` class implements the below methods for the API users to be able to either convert the data to raw-data or store in a file respectively:

- `get_audio_raw()`
- `store_audio()`

So the example should be as below:

```python
if response.error is "":
    # Store audio data to file
    response.store_audio("/tmp/tts.wav")
```

In a similar way you can get the raw-audio-data from the `TextToSpeech.Response` object:

```python
rawAudioData = response.get_raw_data()
```

Will update the documented example.

klpanagi, 2016-05-26:
Updated example in [README.md](https://github.com/rapp-project/rapp-api/blob/devel/python/RappCloud/CloudMsgs/README.md#text-to-speech)

closed 2016-05-26 by klpanagi