rapp-project/rapp-api #28: Python API - Config file where the RAPP Platform IP was declared missing form devel branch

etsardou, 2016-05-26:


athallas, 2016-05-26:
Additionally, when the ip address is replaced with a random, system just deadlocks and connection failure is not returned.

etsardou, 2016-05-31:
@klpanagi is this done? Where is the configuration file stored?

etsardou, 2016-05-31:
Found it. Change ```rapp_platform_python_api``` folder to ```rapp_platform``` which already exists. Furthermore do we have to create the folder by hand or you have changed the setup scripts? Finally, a configuration file sample / template must exist somewhere.

klpanagi, 2016-06-01:
@athallas 
> Additionally, when the ip address is replaced with a random, system just deadlocks and connection failure is not returned.

You investigate this behavior because of the network configuration of the machine executing the calls. This is not an issue of the API implementation. TCP packets do not arrive at destination address (RAPP-Platform) thus no responses are ever received. You can explicitly tell to stop waiting for a response after a given number of seconds with the timeout parameter in `RappPlatformService` constructor ([look here](https://github.com/rapp-project/rapp-api/tree/master/python#calling-rapp-platform-services---api-usage)).

@etsardou 

> Found it. Change rapp_platform_python_api folder to rapp_platform which already exists. Furthermore do we have to create the folder by hand or you have changed the setup scripts? Finally, a configuration file sample / template must exist somewhere.

It has not been documented yet.
The `rapp_platform_python_api` directory was selected because **rapp-platform-python-api** is a client tool/library for accessing **rapp-platform** resources and those can be installed on different machines (also for package distribution purposes). Though i can prepare a patch for the configuration file to be placed under `~/.config/rapp_platform/`.
Regarding the creation of the respective path and default configuration file, i will prepare a patch.

klpanagi, 2016-06-01:
Directory change + default config file creation:  https://github.com/rapp-project/rapp-api/commit/7115cdd74c632e290b2b13ffb1c2d84618db7779

Copy default config file in proper directory: https://github.com/rapp-project/rapp-api/commit/7d890bacd391e91e3744ccb830f374729aaa4fee

Changes in **rapp-platform-scripts** repository, to install the module in normal mode. Required for the above to work properly: https://github.com/rapp-project/rapp-platform-scripts/commit/b548ee63c79a15bf5bb8d86f8e94a89003920494

etsardou, 2016-06-06:
Is this ready to be closed?

klpanagi, 2016-06-06:
Merged [pull-request](https://github.com/rapp-project/rapp-platform-scripts/pull/12) in rapp-platform-scripts repo.

Merged in python-platform-api `devel`

TravisCI builds [passed](https://travis-ci.org/rapp-project/rapp-platform/jobs/135227039#L2577)

Marking this issue as resolved.

commit e80a508e (2016-05-26): Load default platform params from config file - Fixes #28
closed 2016-06-06 by klpanagi