rapp-project/rapp-api #30: Rename rapp-api repository to rapp-platform-api

etsardou, 2016-05-31:


maciek-slon, 2016-05-31:
If this repo is intended to be only platform-api, shouldn't there be yet another repo with RAPP-objects, as many robot-api methods needs them (e.g. picture, face, path)?

etsardou, 2016-05-31:
Since the Robot API is now in the rapp-robot-api repository, this should contain only the Platform API. By RAPP-objects I guess you mean code that is common to both APIs? If yes, we could create another repository but of course this will make things more complicated. Also the documentation must specifically state what to do in case you want to build / install a repo (or we can simply create setup scripts). If you want me to create this repo, tell me so.

maciek-slon, 2016-05-31:
Let's wait a while for other opinions, but for me this is a way to go, as we already decided to split repos. Setup scripts should be rather easy to create for those three repos.

alexge233, 2016-07-10:
@etsardou there are shared classes (under `objects`) across robot and platform. Whilst I do see the point and agree, this will be confusing to say the least.
A better solution would be to add the robot api as extension into the rapp api, by merging them.

closed 2016-07-10 by alexge233