rapp-project/rapp-api #37: add plan_ath_2d API implementation

dudekw, 2016-06-08:
Current plan_path_2d API implementation is copy-paste of a speech API method. Proper plan_path_2d method implementation is required to the incoming release. Could you help me with the issue?

alexge233, 2016-06-08:
Hi,

This is the only API call I haven't implemented yet, I need some kind of wrapper since the documentation was using ROS parameters.

maciek-slon, 2016-06-08:
Look at our branch, wut_devel, where all the necessary objects (pose, pose_stamped, point, path etc.) are created: https://github.com/rapp-project/rapp-api/tree/wut_devel/cpp/includes/objects

dudekw, 2016-06-13:
@alexge233 If you require any further information, let me know.

dudekw, 2016-06-16:
@alexge233 any updates?

alexge233, 2016-06-16:
@dudekw earliest I can work on this is during this weekend, apologies.

dudekw, 2016-06-17:
Ok, thank you @alexge233

dudekw, 2016-06-22:
Sorry @alexge233 for being so impatient, but @klpanagi would like to test Hazard detection RApp and it depends on this service.

alexge233, 2016-06-24:
Hi,

I'm afraid that I am on vacation and away until the 7th of July.
I've updated the code but haven't tested it; I don't have the latest platform, and I could really use an example (under `/examples`) to go with navigation.

Also, I've raised some issues regarding the JSON format with @klpanagi, especially concerning the keys used for `start_pose` and `end_pose`.

The C++ code *should* work, but the example may require careful linking.
Also there is no response processing yet.

alexge233, 2016-07-10:
@dudekw @klpanagi 

Please note that this issue **cannot be closed or solved** for the following reasons:

1. parameters appear to require a `start` and `goal` pose_stamped. 
It makes no sense to know beforehand future `pose_stamped` and especially the `time` or timestamp, or the quaternion. I would suggest that it is changed to accept a starting `pose_stamped` and a goal `coordinate` or `cartesian` or `pose` (see issue #43) without the use of `time` or `header` (also see issue #42 and issue #41).

2. there are no examples of usage, so I have no way of knowing how to test the function, or even if I am using it correctly. I've made an example (https://github.com/rapp-project/rapp-api/blob/cpp/cpp/tests/serialise.cpp) but its just random stuff. I suggest that someone who understands how it works, documents it in detail or writes a thorough example in C++.

3. the JSON used has no keys for "start" and "goal". While this may be acceptable in JS or Python, in C++/Boost it isn't, as it will most certainly be a source of problems. Thus it requires that a key "start" and a key "end" are given, so that I may deserialise it properly with member checks.

Finally, I suggest that some kind of factory class is written which will create automatically, either `pose` or `pose_stamped`. This most likely will be in the robot-api.

*Please use std::chrono* and not platform-specific time functions!

etsardou, 2016-07-11:
1. In my point of view, indeed the web service only needs the coordinates of the start and end pose (x,y,z, yaw ,pitch, roll) as well as the request timestamp, thus the pose_stamped structure is over-engineering.
Nevertheless I propose to implement this as is in order to proceed with the RApp evaluation (cause we are currently late) and restructure it later.

2. @dudekw can provide a working example for you to test

3. @klpanagi ?

Please try to coordinate in order to close this issue as soon as possible. Thanks!

klpanagi, 2016-07-11:
I agree with @etsardou that it is over-engineered to use the pose_stamped structure for this service.

Web service input arguments depends on the relevant ROS Service request message structure.
Maybe @dudekw can modify the ROS Service request message structure, as proposed by @etsardou, in the future, and i will apply the changes to the web service and rapp-platform-api respectively.

Keep in mind that we are currently working on the **Final RApp implementations** and we do not have the time to make those changes, as we will have to patch both the RAPP Platform and the APIs.

> the JSON used has no keys for "start" and "goal". While this may be acceptable in JS or Python, in C++/Boost it isn't, as it will most certainly be a source of problems. Thus it requires that a key "start" and a key "end" are given, so that I may deserialise it properly with member checks.

I do not understand what you mean. The JSON used has the `start` and `goal` keys... Do you mean that those objects do not have static members? Current used start and goal messages  have the structure of a pose_stamped message.

alexge233, 2016-07-11:
Hi,

Maybe I wasn't clear enough:

1. I have no way of knowing the **future** timestamp for the goal pose. Regardless of the over-engineering aspect (to which I also agree) the goal pose will contain either zero timestamp, or some future date. Is this acceptable, or will it create issues?

2. Theres some confusion over the JSON I'm sending and the one I'm receiving.
I will send a JSON which will contain members `start` and `goal` with named keys, ok?

I'll provide an example later this week.

etsardou, 2016-07-11:
About 1: 

You were clear about the future timestamp, thats why I talked about overengineering on using the ROS structs. 
Nevertheless I cannot see how this affects you since its the user (app) that provides the timestamp variable whereas you must just implement the value passing to the ROS service.

dudekw, 2016-07-11:
1. I agree, we should discuss it later.

2. Unfortunately I can provide working example of the ROS service test only. The web service was developed by @klpanagi, however [here](https://github.com/rapp-project/rapp-platform/blob/devel/rapp_web_services/services/path_planning_plan_path_2d/iface_obj.js#L32) you can find client request fields (am I correct @klpanagi?). If I understand rapp_platform web services implementation correctly, start and goal dictionaries should contain analogous fields to the [pose_stamped](https://github.com/rapp-project/rapp-api/blob/wut-cpp-merge/cpp/includes/objects/pose_stamped/pose_stamped.hpp) structure.
exemplary input:
```
Input = {
  "map_name": 'THE_PRESTORED_MAP_NAME',
  "robot_type": 'NAO',
  "algorithm": "dijkstra",
  "start": {header: {seq: 0, stamp: {sec: 0, nsec: 0}, frameid: '/map'},
            pose: {position: {x: 1, y: 0, z: 0}, orientation: {x: 0, y: 0, z: 0, w: 1}}},

  "goal":  {header: {seq: 0, stamp: {sec: 0, nsec: 0}, frameid: '/map'}, 
            pose: {position: {x: 2, y: 2, z: 0}, orientation: {x: 0, y: 0, z: 0, w: 1}}}
}
```
,where ```THE_PRESTORED_MAP_NAME``` is name of an exemplary map. The map files must exist in the following paths: 

``~/rapp_platform_files/maps/"+'user_name'+"/"+'THE_PRESTORED_MAP_NAME'+".yaml``

``~/rapp_platform_files/maps/"+'user_name'+"/"+'THE_PRESTORED_MAP_NAME'+".png``

exemplary map files can be found here: (https://github.com/rapp-project/rapp-platform/tree/master/rapp_path_planning/rapp_map_server/maps)

Yes, you can pass zeros to the goal time stamp.

alexge233, 2016-07-11:
@etsardou it doesn't affect me, other than I will ignore empty or incorrect timestamps (in fact I will do no checks), just asking and making it clear to all parties.

@dudekw thanks for the reply. I will finalise the api, and write a functional example using what you've provided. I will notify you (and close the issue) once I do so, and then ask you to test it.

I am also guessing that I need to do the JSON deserialisation of the planned path?
I don't see anything doing that in the robot-api code.

alexge233, 2016-07-17:
Closing this, I've implemented the class but it is **untested** - somebody with the platform running please test it, and re-open if you find any issues.

Once verified its working, I will migrate the code from `handle_reply` into individual handlers of all `struct` involved, in a `load_from_json` function.

closed 2016-07-17 by alexge233

etsardou, 2016-07-18:
@alexge233 the RAPP Platform is launched in 155.207.19.229, so you can test the call's validity as well.

alexge233, 2016-07-18:
@etsardou this ip is either blocking me, or is down, I only get time-outs.

etsardou, 2016-07-19:
I have just tested it and its working. To be sure you can test it by executing [this](https://github.com/rapp-project/rapp-platform/blob/master/rapp_testing_tools/scripts/rapp_run_test.py). If this does not execure correctly please post the output.

Of course, make sure that you have declared the correct IP/port in the config file.

alexge233, 2016-07-25:
Still can't access it `ping 155.207.19.229` returns timeouts, as does `ssh`.
Do I need to be whitelisted or something?

etsardou, 2016-07-25:
Did you try it with the Python API as well? ssh wont work, its disabled.

If both C++ and Python APIs wont work then something is wrong with the server. @maciek-slon can you try calling a service in this IP as well?

alexge233, 2016-07-25:
No I haven't tried with python, the host seems unreachable from here, are you accessing it from outside your LAN/WAN ?

etsardou, 2016-07-25:
I have just tested it from home (out of the university) using the Python API (I just executed the ```rapp_run_test.py``` and I can access it. When you say time-out what do you mean? Can you post the output? Also the credentials in the ```.config/rapp_platform``` are correct?

alexge233, 2016-07-25:
@etsardou time-out as in the server does not respond to me (simple pings).

*EDIT* I was wrong, its simply taking too long to respond, and I am guessing the connections are assumed dropped!

```
Starting Nmap 6.49BETA5 ( https://nmap.org ) at 2016-07-25 17:06 BST
Nmap scan report for rapp.ee.auth.gr (155.207.19.229)
Host is up (0.12s latency).
Not shown: 999 filtered ports
PORT     STATE SERVICE
9001/tcp open  tor-orport

Nmap done: 1 IP address (1 host up) scanned in 92.66 seconds
```

I did a traceroute as well, it seems that on average its more than 120sec to access the platform :-(

etsardou, 2016-07-25:
I think ping is disabled as well. Can you run the python script?

alexge233, 2016-07-25:
no because it requires the platform.
I'm trying to solve the `broken pipe` error I get, which I think it has to do with the header.

etsardou, 2016-07-25:
You mean you have not cloned the platform in your PC? It does not require the platform to be running.

alexge233, 2016-07-25:
ok, I'll give it a try then!
Does the token need to be valid at this point?

etsardou, 2016-07-25:
Of course. You can execute this to setup the token https://github.com/rapp-project/rapp-platform-scripts/blob/master/setup/13_authentication_setup.sh

alexge233, 2016-07-25:
ok, this explains the `401` (invalid http header) which in reality translates to "Unauthorized access".
it does not however solve the `broken pipe` issue.

dudekw, 2016-08-08:
@alexge233  Is this issue resolved? I tried to compile the path_planning service example and I got some errors in `rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp` file.

alexge233, 2016-08-08:
@dudekw can you post the errors please?

dudekw, 2016-08-08:
after including:
``` cpp
#include "objects/pose_stamped/pose_stamped.hpp"
#include "objects/planned_path/planned_path.hpp"
#include "objects/yaml/yaml.hpp"
#include "objects/picture/picture.hpp"
#include <chrono>
#include <boost/property_tree/json_parser.hpp> 
#include "objects/pose_metadata/pose_metadata.hpp"
```
and changing:
``` cpp
 tree.add_child("", end.treefy());
```
to:
``` cpp
 tree.add_child("", goal.treefy());
```
after above changes, got following erorrs: 

```
In file included from /home/wojciech/rapp/rapp-api/src/rapp-api/cpp/examples/path_planning.cpp:2:0:
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp: In member function ‘void rapp::cloud::plan_path_2d::handle_reply(std::string)’:
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:94:58: error: expected primary-expression before ‘(’ token
       meta = std::unique_ptr<rapp::object::pose_metadata>(new meta(iter));
                                                          ^
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:94:63: error: expected type-specifier before ‘meta’
       meta = std::unique_ptr<rapp::object::pose_metadata>(new meta(iter));
                                                               ^
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:97:22: error: ‘pose’ was not declared in this scope
      std::unique_ptr<pose> pose;
                      ^
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:97:22: note: suggested alternative:
In file included from /home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/objects/pose_stamped/includes.ihh:1:0,
                 from /home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/objects/pose_stamped/pose_stamped.hpp:3,
                 from /home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:5,
                 from /home/wojciech/rapp/rapp-api/src/rapp-api/cpp/examples/path_planning.cpp:2:
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/objects/pose/pose.hpp:13:8: note:   ‘rapp::object::pose’
 struct pose
        ^
In file included from /home/wojciech/rapp/rapp-api/src/rapp-api/cpp/examples/path_planning.cpp:2:0:
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:97:26: error: template argument 1 is invalid
      std::unique_ptr<pose> pose;
                          ^
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:97:26: error: template argument 2 is invalid
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:97:32: error: invalid type in declaration before ‘;’ token
      std::unique_ptr<pose> pose;
                                ^
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:98:21: error: ‘else’ without a previous ‘if’
                     else if (iter->first == "pose") {
                     ^
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:99:30: error: the value of ‘pose’ is not usable in a constant expression
       pose = std::unique_ptr<pose>(new pose(iter));   
                              ^
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:97:22: note: ‘pose’ was not declared ‘constexpr’
      std::unique_ptr<pose> pose;
                      ^
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:99:34: error: type/value mismatch at argument 1 in template parameter list for ‘template<class _Tp, class _Dp> class std::unique_ptr’
       pose = std::unique_ptr<pose>(new pose(iter));   
                                  ^
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:99:34: error:   expected a type, got ‘pose’
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:99:34: error: template argument 2 is invalid
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:99:40: error: expected type-specifier before ‘pose’
       pose = std::unique_ptr<pose>(new pose(iter));   
                                        ^
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:102:39: error: expected primary-expression before ‘:’ token
                     path.push_back(std:move(ps));
                                       ^
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/cloud/navigation/path_planning/path_planning.hpp:111:84: error: no matching function for call to ‘rapp::object::planned_path::planned_path(std::string&, std::string&, std::vector<rapp::object::pose_stamped>&)’
         delegate_(std::move(rapp::object::planned_path(plan_found, plan_error, path)));
[... 39 lines cut ...]
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/examples/path_planning.cpp:30:86: error: no matching function for call to ‘rapp::object::pose_metadata::pose_metadata(int, std::chrono::nanoseconds&)’
[... cut ...]
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/objects/pose_metadata/pose_metadata.hpp:30:5: note:   candidate expects 0 arguments, 2 provided
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/objects/pose_metadata/pose_metadata.hpp:21:5: note: rapp::object::pose_metadata::pose_metadata(int, rapp::object::time, std::string)
     pose_metadata(
     ^
/home/wojciech/rapp/rapp-api/src/rapp-api/cpp/includes/objects/pose_metadata/pose_metadata.hpp:21:5: note:   candidate expects 3 arguments, 2 provided
```

alexge233, 2016-08-08:
@dudekw can you post your example here on Github, or submit it in branch `cpp` under `examples` ?
I think its only a matter of using the namespace (I forgot `rap::object::`) but I would like to test it myself.
Many thanks.

dudekw, 2016-08-08:
Following example should do the work:

```cpp
#include "cloud/service_controller/service_controller.hpp"
#include "cloud/navigation/path_planning/path_planning.hpp"
#include "objects/pose_stamped/pose_stamped.hpp"
#include "objects/planned_path/planned_path.hpp"
#include "objects/yaml/yaml.hpp"
#include "objects/picture/picture.hpp"
#include <chrono>
#include <stdlib.h>
///
/// Pass as param start and goal poses and path to png and yaml file and map name
///
int main(int argc, char* argv[])
{
    if (argc == 18) {
  	// service controler
        rapp::cloud::platform_info info = {"155.207.19.229", "9001", "rapp_token"}; 
	rapp::cloud::service_controller ctrl(info);

	// callback lambda UPLOAD MAP
	auto upload_callback = [&](std::string status)
						{std::cout << "upload status: \n" << status << " \n" << std::endl;};
	// upload map
        std::string yaml_file_name  = argv[16];
	std::string png_file_name  = argv[15];
	auto yaml_file = rapp::object::yaml(yaml_file_name);
	auto picture_file = rapp::object::picture(png_file_name);
	ctrl.make_call<rapp::cloud::path_upload_map>(picture_file, yaml_file, upload_callback);

        //set start and goal 
	std::chrono::nanoseconds nanosec(10);
        auto start = rapp::object::pose_stamped(rapp::object::pose_metadata(0,nanosec),rapp::object::pose(rapp::object::point((float) atof(argv[1]),(float) atof(argv[2]),(float) atof(argv[3])), rapp::object::quaternion((float) atof(argv[4]),(float) atof(argv[5]),(float) atof(argv[6]),(float) atof(argv[7]))));

        auto goal = rapp::object::pose_stamped(rapp::object::pose_metadata(0,nanosec),rapp::object::pose(rapp::object::point((float) atof(argv[8]),(float) atof(argv[9]),(float) atof(argv[10])), rapp::object::quaternion((float) atof(argv[11]),(float) atof(argv[12]),(float) atof(argv[13]),(float) atof(argv[14]))));

	// callback lambda PLAN_PATH_2D
	auto callback = [&](rapp::object::planned_path path)
						{std::cout << "is plan found? \n" << path.plan_found << " \n" << std::endl;};

	// detect faces
        std::string map_name  = argv[17];
	ctrl.make_call<rapp::cloud::path_planning>(map_name, "NAO", start, goal, callback);
	return 0;
    }
}
```

alexge233, 2016-08-08:
@dudekw  a few remarks:

1. you're constructing a `pose_metadata` using `std::chrono::nanoseconds`, but it takes a parameter `sequence`, `time` and `frameid`.
2. `rapp::object::time` takes only one parameter, not two.
3. there is no cloud call `path_planning`, theres only `plan_path_2d` and `path_upload_map`. My guess is you want the first, which also takes an additional parameter `algorithm`.
4. similarly your first call to `path_upload_map` is missing the string parameter `map`.

I've uploaded an example `examples/path_planner.cpp` which compiles but may not work (haven't tested it yet).

dudekw, 2016-08-08:
@alexge233 Thank you very much! Sorry for untested example code, but I couldn't compile it before.