rapp-project/rapp-api #41: [cpp] rapp::object::time has erroneous logic and usage

alexge233, 2016-07-10:
@dudekw The implementation of `rapp::object::time` is dangerous and flawed:

1. it requires bot seconds and nanoseconds (no need, either should suffice)
2. it uses `int32_t`

I suggest it becomes a wrapper around either `std::crhono` or `boost::posix_time`
In the `test` directory you can see how I instantiate it:
https://github.com/rapp-project/rapp-api/blob/cpp/cpp/tests/serialise.cpp

Furthermore, implicit casting to `int32_t` is tricky and potentially very dangerous.

alexge233, 2016-07-24:
**UPDATE** the type `uint32_t` (e.g., `unsigned int`) is **not long enough** to hold UNIX epoch in nanoseconds!!!

Example using `now()` with uint32_t:
(seconds and nanoseconds)
```
1469376406 - 1694538216
```

Example using `now()` with uint32_t and uint64_t (aka `unsigned long long`)

```
1469376489 - 1469376489060072000
```

For this reason I have reverted to using:

`std::chrono::system_clock::time_point`

as `class time` private member from which I cast using `std::chrono::duration_cast<std::chrono::seconds>` or `<std::chrono::nanoseconds>`, which also has the added benefit of adjusting the clock to time changes.

closed 2016-07-24 by alexge233

maciek-slon, 2016-07-25:
I was away for some time and just noticed whole discussion. To clarify things with time struct:
- `seconds` holds whole seconds part of time point
- `nanoseconds` holds (or should hold) only the fractional part of whole seconds, thus this field is in range [0..1e9)

At least that is what ROS-related algorithms are expecting as an input data.

alexge233, 2016-07-25:
Its a good thing you told me, I have to refactor the class, and subtract nanoseconds from seconds.