rapp-project/rapp-api #44: Update master branch for release

etsardou, 2016-07-12:
- [ ] Merge C++ work
- [ ] Merge JS work (if needed)
- [ ] Update README.md file with the implementations
- [ ] @klpanagi Add important links in README.md such as:

- https://github.com/rapp-project/rapp-api/blob/master/python/README.md
- https://github.com/rapp-project/rapp-api/blob/master/python/RappCloud/README.md
- https://github.com/rapp-project/rapp-api/blob/master/python/RappCloud/CloudMsgs/README.md

alexge233, 2016-07-13:
@etsardou @dudekw I have rejected the pull request; there are major conflicts, and also some classes must be refactored.

klpanagi, 2016-11-20:
Proper README.md links exist on branch **devel**.

alexge233, 2016-11-21:
@etsardou and @klpanagi I've updated master with `cpp` branch, and merged `cpp_dev` on `devel`.
However, I personally think that master should have the `cpp_dev` / `devel` code since it has been extensively tested with unit tests and stress tests. Let me know if you want me to proceed with that.

klpanagi, 2016-11-21:
@alexge233, I really cannot read the git tree anymore...

You have made a direct merge to **master** (even a direct commit to master) and the tree has been broken. Even **devel** branch holds a direct commit!

**devel** branch holds both **js** and **python** development work and should be the one to merge into **master**.

If i understand correctly, **devel** **CANNOT** be merged to **master** because of conflicts in cpp implementations...

alexge233, 2016-11-21:
direct commit was a direct edit of README from github.
devel branch has had cpp_dev commited because I'm removing it tomorrow.
Only file in conflict was README.md
Any conflict in cpp implementation is solved by accepting devel/cpp_dev and ignoring master (if we are to move to 0.7.0)

alexge233, 2017-02-06:
C++ is merged and clear.