rapp-project/rapp-api #47: [cpp] API example face_detect not working

maciek-slon, 2016-07-25:
I've tried to run example `face_detect` application from both `devel` and `cpp` branch. I have platform compiled today, from the setup scripts, on clean machine. When running python tests, web service shows this output on terminal:

```
40 ACCEPT: localhost [Mon, 25 Jul 2016 15:26:01 +0200]
40 REQUEST.serv: POST *://anonymous@localhost:9001 /hop/face_detection HTTP/1.1
40 INVOKE.svc (face_detection . #|JsObject [__proto__: JsObject] [extensible: #t] [properties: ()] [cmap: JsConstructMap] [elements: vector]|)
```

But when running C++ example, output is a little bit different:

```
41 ACCEPT: localhost [Mon, 25 Jul 2016 15:37:07 +0200]
41 REQUEST.serv: POST *://anonymous@localhost:9001 /hop/face_detection HTTP/1.1
41 INVOKE.svc (face_detection)
```

Output from the `face_detect` is:

```
face detect: /home/rapp/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform/rapp_testing_tools/test_data/face_samples/etsardou_close_angle.jpg
found 0 faces!
```

And application is not finishing, it hangs instead (must be killed by CTRL+C).

Given test file, when passed directly to ROS service (by `rosservice` call), produces expected output (one face).

alexge233, 2016-07-25:
We need a public platform on which we can test, I have no platform running atm so I can't test this.
From the looks of it, the JSON param isn't being sent, or isn't being parsed properly.

etsardou, 2016-07-25:
https://github.com/rapp-project/rapp-api/issues/37#issuecomment-233591379

alexge233, 2016-07-25:
If the application hangs, then my guess is the connection is not closed properly.
Do you get a time-out at all?

Can you please print the entire output from the console, and add the following line
in `cpp/includes/cloud/asio_service_http/asio_service_http.cpp:23`

```cpp
std::cout << header_ << post_;
```

maciek-slon, 2016-07-25:
I,ve waited for few (10 maybe?) seconds before pressing CTRL+C and there was no timeout.

> Can you please print the entire output from the console, and add the following line
> in cpp/includes/cloud/asio_service_http/asio_service_http.cpp:23

I will do this tomorrow morning, even before morning coffee ;-)

alexge233, 2016-07-25:
Well in my case `broken pipe` is the error due to the server closing the connection AFAIK.
You said this is a new platform install, have you tested it with Python and found it working?

etsardou, 2016-07-25:
PS. You can also try it locally using the latest Platform OVA: http://rapp-project.eu/wp-content/uploads/2016/07/rapp-platform-0.6.ova_.tar

alexge233, 2016-07-25:
@etsardou my little laptop won't run the platform in a VM, I'll have to test with a desktop but because we're moving offices this week it will be hard.

etsardou, 2016-07-25:
A general comment: The C++ API must be finished ASAP as we are later than late in the hazard detection application testing (which is developed in C++).

alexge233, 2016-07-25:
it is complete, with the exception of bug fixes as they arise (such as `face_detection`). I'm refactoring `struct time` atm.

alexge233, 2016-07-26:
Hi all,

Commit eb77c0067405f781b001c2a00d267724110272d2 has resolved *some* of the issues.
Due to changing the HTTP header, it was being created as a malformed header.
Also, if you don't use a *correct* authentication token, you will receive error 401, which although misleading, means "no access".
However, as of now, NONE of the C++ API calls seem to communicate data to the server setup by AUTH at 155.207.19.229.

I don't know why that is, the HTTP Header and POST seem fine, and the data is being transmitted AFAIK.

I tested `face_detection`, `qr_detection` and `ontologies`, and I got the following:

```
POST /hop/ontology_subclasses_of HTTP/1.1
Host: 155.207.19.229
Accept-Token: rapp_token
Connection: close
Content-Length: 41

{"ontology_class":"car","recursive":"0"}
{"results":[],"error":"Error, empty ontology class"}
```

For qr:

```
POST /hop/qr_detection HTTP/1.1
Host: 155.207.19.229
Accept-Token: rapp_token
Connection: close
Content-Length: 691
Content-Type: multipart/form-data; boundary=SbMohei8RyHa2ffF

--SbMohei8RyHa2ffF
Content-Disposition: form-data; name="json"

{"file":"IJZMXMJFb16W3rRH.png"}

--SbMohei8RyHa2ffF
Content-Disposition: form-data; name="file"; filename="IJZMXMJFb16W3rRH.png"
Content-Type: image/png
Content-Transfer-Encoding: binary

�PNG

IHDRJJAPLTE���U~MIDA��0
��ތ�@�P������vڥ�sTm&Zn��)vV�AUQA�ѬE[Er��ZYP�7$ng\�6$��ћ��]��[�G\@�P[���PI��6F3[�9ط
�t�����ŗ�Hg�x��
               �7�=e�*�����]�z�ؙ͟UAǧ0CsH��(jv�
                                            Aurz-ڤ���Z@gh��w��bh�Bo%�>IEND�B`�
--SbMohei8RyHa2ffF--{"qr_centers":[],"qr_messages":[],"error":"No image file received"}
found 0 QR codes
```

Same story with face_detect, and I am guessing it will be the same story with the rest.

etsardou, 2016-07-26:
Actually these responses means that data are communicated. 

For the ontology you have an error ```Error, empty ontology class```, meaning that this class does not exist in the ontology (which clearly is a bug since it must have returned "Not a valid ontology class" @akintsakis ). Try giving "Oven" (with capital O).

For the QR the request has happened (that's why you got a response with an error) and it seems that the data you provided were not correct. @klpanagi do you have any idea why?

Again the Python API works fine, so the web services are most probably correct.

maciek-slon, 2016-07-26:
As you already managed to print requests made from C++, I've printed those made by python calls. Sample request made from python looks like this:

```
POST http://localhost:9001/hop/face_detection
user-agent: rapp-platform-api/python
Content-Length: 474107
Content-Type: multipart/form-data; boundary=fe646a5ce45d4e0a8ca2b3627e10562c
Accept-Token: rapp_token

--fe646a5ce45d4e0a8ca2b3627e10562c^M
Content-Disposition: form-data; name="json"^M
^M
{"fast": true}^M
--fe646a5ce45d4e0a8ca2b3627e10562c^M
Content-Disposition: form-data; name="file"; filename="Lenna-76m42AgfqRnT4REdlEOmckd4qweDtX..png"^M
^M
<89>PNG^M
^Z
^@^@^@^MIHDR^@^@^B^@^@^@^B^@^H^B^@^@^@{^ZC [rest of the PNG file]
--fe646a5ce45d4e0a8ca2b3627e10562c--
```

klpanagi, 2016-07-26:
@alexge233 ,

the way you POST data (parameters and files) is incorrect according to the HOP Server specifications.
Look [here](https://github.com/rapp-project/rapp-platform/tree/master/rapp_web_services/services#service-specifications---request-arguments-and-response-objects) for more information and examples of valid POST requests

klpanagi, 2016-07-26:
**POST application/x-www-urlencoded - JSON stringify data**:

```bash
curl -v -X POST -d "json=%7B%22ontology_class%22%3A+%22Oven%22%2C+%22recursive%22%3A+false%7D" -H "Accept-Token:rapp_token" 155.207.19.229:9001/hop/ontology_subclasses_of
```

```http
POST /hop/ontology_subclasses_of HTTP/1.1
User-Agent: curl/7.35.0
Host: 155.207.19.229:9001
Accept: */*
Accept-Token:rapp_token
Content-Length: 73
Content-Type: application/x-www-form-urlencoded

json=%7B%22ontology_class%22%3A+%22Oven%22%2C+%22recursive%22%3A+false%7D

HTTP/1.1 200 Ok
Content-Length: 170
Connection: keep-alive
Content-type: application/json; charset=UTF-8
Server: Hop

{"results":["http://knowrob.org/kb/knowrob.owl#MicrowaveOven","http://knowrob.org/kb/knowrob.owl#RegularOven","http://knowrob.org/kb/knowrob.owl#ToasterOven"],"error":""}

````

alexge233, 2016-07-26:
Now I am really confused, this code was working during the last hackathon, what changed in HOP now?

@klpanagi where is the error? 

1. I've tried various combinations with the header lines, and I still get the error that the file was not uploaded.

2. I do not send `application/json` requests, only `multipart/form-data`

alexge233, 2016-07-26:
@mszlenk  the POST field you've shown from Python is incorrect:

`POST http://localhost:9001/hop/face_detection`

How is this working with HOP?

alexge233, 2016-07-26:
Here are the headers and POST from ontology calls:

````
POST /hop/ontology_subclasses_of HTTP/1.1
Host: 155.207.19.229
Connection: close
Content-Length: 42
Accept-Token: rapp_token

{"ontology_class":"Oven","recursive":"0"}
{"results":[],"error":"Error, empty ontology class"}
```

and

```
POST /hop/ontology_superclasses_of HTTP/1.1
Host: 155.207.19.229
Connection: close
Content-Length: 42
Accept-Token: rapp_token

{"ontology_class":"Oven","recursive":"0"}
{"results":[],"error":"Error, empty ontology class"}
```

And this is the qr detection, @klpanagi would you please indicate what is the bug?

```
POST /hop/qr_detection HTTP/1.1
Host: 155.207.19.229:9001
User-Agent: rapp_api/cpp/0.6.0
Accept: */*
Accept-Token: rapp_token
Content-Length: 689
Content-Type: multipart/form-data; boundary=wAAcF3eZuUSyMZqL

--wAAcF3eZuUSyMZqL
Content-Disposition: form-data; name="json"

{"file":"XOj2yjZ0W8iKnSYB.png"}
--wAAcF3eZuUSyMZqL
Content-Disposition: form-data; name="file"; filename="XOj2yjZ0W8iKnSYB.png"
Content-Type: image/png
Content-Transfer-Encoding: binary

�PNG

IHDRJJAPLTE���U~MIDA��0
��ތ�@�P������vڥ�sTm&Zn��)vV�AUQA�ѬE[Er��ZYP�7$ng\�6$��ћ��]��[�G\@�P[���PI��6F3[�9ط
�t�����ŗ�Hg�x��
               �7�=e�*�����]�z�ؙ͟UAǧ0CsH��(jv�
                                            Aurz-ڤ���Z@gh��w��bh�Bo%�>IEND�B`�
--wAAcF3eZuUSyMZqL--
{"qr_centers":[],"qr_messages":[],"error":"No image file received"}
````

The picture is valid, I have hexdumped to verify.

klpanagi, 2016-07-27:
Nothing has changed about the HOP Server

```http
POST /hop/ontology_subclasses_of HTTP/1.1
Host: 155.207.19.229
Connection: close
Content-Length: 42
Accept-Token: rapp_token

{"ontology_class":"Oven","recursive":"0"}
{"results":[],"error":"Error, empty ontology class"}
```

What is the **Content-Type** of this POST request? Format of the payload indicates that the data are in `application/json` format (`{"ontology_class":"Oven","recursive":"0"}`) which is not supported by the HOP Server.

Data have to be serialized to a JSON string as the value of the parameter `json` as described [here](https://github.com/rapp-project/rapp-platform/tree/master/rapp_web_services/services#service-specifications---request-arguments-and-response-objects)
Thus the POST should look like this in case of `application/x-www-urlencoded:

```http
POST /hop/ontology_subclasses_of HTTP/1.1
User-Agent: curl/7.35.0
Host: 155.207.19.229:9001
Accept: */*
Accept-Token:rapp_token
Content-Length: 73
Content-Type: application/x-www-form-urlencoded

json=%7B%22ontology_class%22%3A+%22Oven%22%2C+%22recursive%22%3A+false%7D
```

Regarding the `multipart/form-data` POSTS, for Web Services that require file uploads, you have an extra `Content-Disposition: form-data` where you provide the **file** and that is incorrect and it conflicts with the file upload form-data. Try removing that extra form-data and it should work. The POST in that case should look like below:

```http
POST /hop/qr_detection HTTP/1.1
Host: 155.207.19.229:9001
User-Agent: rapp_api/cpp/0.6.0
Accept: */*
Accept-Token: rapp_token
Content-Length: 689
Content-Type: multipart/form-data; boundary=wAAcF3eZuUSyMZqL

--wAAcF3eZuUSyMZqL
Content-Disposition: form-data; name="file"; filename="XOj2yjZ0W8iKnSYB.png"
```

alexge233, 2016-07-27:
1. There doesn't exist a `JSON.stringify` method in c++, so I have to roll out my own. Until then `cpp` branch is non-functional.

2. Why was the API reverted from strings to JSON, if hop doesn't support JSON? 
We had a functional API and now we don't.

3. Where's the extra boundary? 
One for the JSON, one for the BINARY and one to close off the POST, that is the RFC standard.
The description of [qr_detection](https://github.com/rapp-project/rapp-platform/tree/master/rapp_web_services/services#qr-detection) has a JSON "{ file: '' }" so, one boundary for the JSON and one for the binary, what you have shown above is that there is no json `{ file: '' }` although the documentation says there is.

klpanagi, 2016-07-27:
@alexge233 ,

we agreed on this specifications a few months back...

2.
> Why was the API reverted from strings to JSON, if hop doesn't support JSON? 
We had a functional API and now we don't.

Both `js` ( @lpenteri   ) and `python` APIs are functional...
HOP doesn't support parsing `application/json` content type. This does not mean that we cannot send data in json format... We dont have access to the HOP Server implementation so we could not do any modifications like apply an `application/json` parser on incoming requests.

1.

> There doesn't exist a JSON.stringify method in c++, so I have to roll out my own. Until then cpp branch is non-functional.

For the **qr-detection** example you provided, you actually send a  JSON:

```http
POST /hop/qr_detection HTTP/1.1
Host: 155.207.19.229:9001
User-Agent: rapp_api/cpp/0.6.0
Accept: */*
Accept-Token: rapp_token
Content-Length: 689
Content-Type: multipart/form-data; boundary=wAAcF3eZuUSyMZqL

--wAAcF3eZuUSyMZqL
Content-Disposition: form-data; name="json"

{"file":"XOj2yjZ0W8iKnSYB.png"}
```

So if you dont want to make an `application/x-www-urlencoded` POST you can still do the POST in `multipart/form-data` . Below are examples of doing so for the **ontology_subclasses_of** service using `curl cli`. Both  `application/x-www-urlencoded` and `multipart/form-

#### application/x-www-urlencoded

** Command**:

```bash
curl -X POST -d 'json={"ontology_class":"Oven"}' -H "Accept-Token:rapp_token" 155.207.19.229:9001/hop/ontology_subclasses_of
```

**POST request**:

```http
POST /hop/ontology_subclasses_of HTTP/1.1
User-Agent: curl/7.35.0
Host: 155.207.19.229:9001
Accept: */*
Accept-Token:rapp_token
Content-Length: 30
Content-Type: application/x-www-form-urlencoded

json={"ontology_class":"Oven"}
```

**Server Response**:

```http
Content-Length: 170
Connection: keep-alive
Content-type: application/json; charset=UTF-8
Server: Hop

{"results":["http://knowrob.org/kb/knowrob.owl#MicrowaveOven","http://knowrob.org/kb/knowrob.owl#RegularOven","http://knowrob.org/kb/knowrob.owl#ToasterOven"],"error":""}
```

#### application/x-www-urlencoded

** Command**:

```bash
curl -X POST -F 'json={"ontology_class":"Oven"}' -H "Accept-Token:rapp_token" 155.207.19.229:9001/hop/ontology_subclasses_of
```

**POST request**:

```http
OST /hop/ontology_subclasses_of HTTP/1.1
User-Agent: curl/7.35.0
Host: 155.207.19.229:9001
Accept: */*
Accept-Token:rapp_token
Content-Length: 164
Expect: 100-continue
Content-Type: multipart/form-data; boundary=------------------------46bccfd31c49fc3c

--------------------------46bccfd31c49fc3c
Content-Disposition: form-data; name="json"

{"ontology_class":"Oven"}
--------------------------46bccfd31c49fc3c--
```

**Server Response**:

```http
Content-Length: 170
Connection: keep-alive
Content-type: application/json; charset=UTF-8
Server: Hop

{"results":["http://knowrob.org/kb/knowrob.owl#MicrowaveOven","http://knowrob.org/kb/knowrob.owl#RegularOven","http://knowrob.org/kb/knowrob.owl#ToasterOven"],"error":""}

```

3.

> Where's the extra boundary? 
One for the JSON, one for the BINARY and one to close off the POST, that is the RFC standard.
The description of qr_detection has a JSON { file: '' } so, one boundary for the JSON and one for the binary, what you have shown above is that there is no JSON { file: '' } although the documentation says there is.

According to [RFC-1867](https://www.ietf.org/rfc/rfc1867.txt), the **filename** may be described using the `filename` parameter of the **content-disposition** header, thus the boundary for the JSON is useless. The name of the file is optional and **should** be supplied as part of the **content-disposition** header. You anyway, provide the `filename` as a part of the binary boundary so what is the use of applying it also in a separate boundary??

alexge233, 2016-07-27:
@klpanagi 

In `qr_detection` (and similarly in all other calls) what I send is **both** a JSON **and** a binary.

The documentation you have for [qr_detection](https://github.com/rapp-project/rapp-platform/tree/master/rapp_web_services/services#qr-detection) states it takes as param:

```
{ file: '' }
```

That implies that you require the filename in a JSON.

Same with `face_detection` the documentation [states:](https://github.com/rapp-project/rapp-platform/tree/master/rapp_web_services/services#face-detection)

```
{ file: '', fast: false }
```

So, if you check the code, again I send both a JSON and a BINARY.
That is the reason for the extra boundary: one for the JSON and one for the BINARY.

If you don't want me to send the JSON its very easy for me to simply post BINARY data with the filename in the `Content-Disposition` as you suggested.

1. *Will this solve the issue of images not being uploaded?*

2. *Should I remove* the JSON descriptors for the filenames from all cloud calls, and leave **only** the filename descriptor in the binary `Content-Disposition` boundary?

3. Instead of implementing `JSON.stringify` *should I refactor* all calls to `multipart/form-data`?

klpanagi, 2016-07-27:
The documentation states the Web Service input arguments. Thus in case of uploaing files the `{file: ''}` corresponds to the multipart/form-data boundary that is named `file` and you have it correct there.

```http
Content-Disposition: form-data; name="file"; filename="XOj2yjZ0W8iKnSYB.png"
```

The above indicates to the server that param `file` is a binary file and by also supplying `filename` parameter you force the server to use that value.

> If you don't want me to send the JSON its very easy for me to simply post BINARY data with the filename in the Content-Disposition as you suggested.

Yes please.

>  Will this solve the issue of images not being uploaded?

It will solve those issues (check the examples i provided using `curl`)

> Should I remove the JSON descriptors for the filenames from all cloud calls, and leave only the filename descriptor in the binary Content-Disposition boundary?

Yes please.

> Instead of implementing JSON.stringify should I refactor all calls to multipart/form-data?

I dont see the difference and the problem of crafting the payload when using `application/x-www-urlencoded` or `multipart/form-data`

```http
Content-Type: application/x-www-form-urlencoded

json={"ontology_class":"Oven"}
```

**VS**

```http
Content-Disposition: form-data; name="json"

{"ontology_class":"Oven"}
```

But yes if implementing all calls in `multipart/form-data` suits you, that is and wont be a problem

alexge233, 2016-07-27:
@klpanagi please see updated/edited post above.

klpanagi, 2016-07-27:
I have tested both calls to the `ontology_subclasses_of` and `face_detection` from command-line using `curl` and no problems reported... Have a look at the examples i provided.

alexge233, 2016-07-28:
@klpanagi thank you for all your help so far!

I've managed to make them all work, unfortunately there are still issues:

# Ontology

* have you have removed the `recursive` parameter? It works without it, 
* boost does not conform 100% to JSON. it translates everything to strings.
* for example, when I add the `recursive` parameter, I get:

```
{"results":[],"error":"rapp_platform_ros_communications/ontologySubSuperClassesOfSrvRequest message requires a bool for field recursive, but got a <type 'unicode'>"}
```

when I send this:

```
{"ontology_class":"Oven","recursive":"true"}
```

I can manually hack into the strings, but this is an issue, because the same problem will apply to integers and floats. It implies I have to type-cast from strings to write a json (the opposite is ok).
I would rather avoid converting strings to pdt.

# Face detection

1. first I tried curl: `curl -v -X POST -F "file=@Lenna.png" -H "Accept-Token:rapp_token" 155.207.19.229:9001/hop/face_detection`
which worked. Note *no fast params*.

2. then I tried curl: `curl -v -X POST -F "fast=true" -F "file=@Lenna.png" -H "Accept-Token:rapp_token" 155.207.19.229:9001/hop/face_detection`
which produced error: 
`{"faces":[],"error":"rapp_platform_ros_communications/FaceDetectionRosSrvRequest message requires a bool for field fast, but got a <type 'unicode'>"}`

This is the same as with above issue.
It also applies to the C++ API.
Also again if I remove the "fast" param, the call works !

# JSON and PDT
Boost only uses strings for JSON (not 100% compliant).
The C++ call atm hacks into the string to reduce the quoted bool to a literal:

```cpp
    boost::property_tree::ptree tree;
    tree.put("fast", fast);
    std::stringstream ss;
    boost::property_tree::write_json(ss, tree, false);
		
    // set the `fast` param
    post_  = "--" + boundary + "\r\n"
               + "Content-Disposition: form-data; name=\"json\"\r\n\r\n";

    // remove quotes from `true` or `false`
    std::string str = ss.str();
    boost::replace_all(str, "\"false\"", "false");
    boost::replace_all(str, "\"true\"", "true");
    post_ += str;
```

* I can either try to hack a method that removes quotes from all those types
* Extend boost so that it supports PDT (would rather avoid this)
* We can agree only to use strings which are casted internally to their correct PDT

etsardou, 2016-07-28:
Are there comments missing?

klpanagi, 2016-07-28:
@alexge233 ,

About the unicode errors refer to https://github.com/rapp-project/rapp-api/issues/47#issuecomment-235672846

You must send a json.

alexge233, 2016-07-29:
@klpanagi You have inconcistencies:

One JSON uses a *string* boolean:

1. 
```json
{"ontology_class":"Oven","recursive":"0"}
```

2.
And another uses a PDT in the JSON
```json
{"ontology_class":"Oven","recursive":true}
```

Please avoid using PDT in JSON as I have no native support.
This means, please avoid using primitive datatypes, but rather cast them from strings, e.g. the first rather than the latter.

This has nothing to do with POST data.

Many thanks!

closed 2016-07-29 by alexge233

klpanagi, 2016-07-29:
I never referred to **string boolean** in a JSON...
All the examples use PDT in a JSON.
Again, we agreed on this specifications for data transfers using POST verb and both `js` and `python` API implementations use this approach with no problem at all.

We cannot support **only strings** in a JSON as both `js` and `python` APIs as long as the RAPP Platform Web Services have been implemented this way.
This should me mentioned a few months back...

Sorry and thank you!

alexge233, 2016-07-29:
Hi, 

I've made changes to the boolean calls so that they use PDT, but I am telling you right now that the object structs created by WUT which I am now incorporating  into `0.6.0` **cannot use** json pdt.

For simple use-cases (bool, ints) I have removed the quotes from the final JSON string but JSON serialisation in boost/c++ saves everything as strings. 

The `nsec`, `sec`, `point`, `quaternion`, `pose` and `pose_stamped` **can only use** strings which you have to cast to the appropriate type.
There is nothing I can do about this other than bring in a new JSON library dependency, which won't happen for 0.6.0.

Best regards!

etsardou, 2016-07-29:
Since this is what we agreed upon, introduce the JSON library dependency in a future version.

etsardou, 2016-07-29:
Just to be clear, meanwhile do whatever is needed for all the remaining calls to be functional ASAP (e.g. the path planning calls which you mentioned above).