rapp-project/rapp-api #49: [js] API tests not working

maciek-slon, 2016-07-27:
When running `npm test` in `js` subdirectory, following output is produced (all errors looks similar):

```
  #ontology_superclasses_of()
{ [Error: socket hang up] code: 'ECONNRESET' }
```

For this call, web service prints following output:

```
1 ACCEPT: localhost [Wed, 27 Jul 2016 14:07:29 +0200]
1 ABORT:  &hop-method-error
*** ERROR:request-line-grammar:
Method not implemented -- Xv�);��V�rW�,h�f1W�Yt���9��J���/�+�0�,��'g�(k�$��
���jih9876�2�.�*�&���=5�#��	���@?>3210�1�-�)�%���</�
�f
   	localhost
                 


	
 
    1. &pool-scheduler1070, 
```

API branch: `devel`
Platform branch: `devel`

klpanagi, 2016-07-27:
That is due to the fact that the `js` API uses `https` as the default protocol and the RAPP Platform is launched at `http`  ( [here?](https://github.com/rapp-project/rapp-api/blob/devel/js/includes/cloud/RAPPCloud.js#L11) )

**Note**: The configuration parameters regarding RAPP Platform host should be read from a configuration file, similar to the [python](https://github.com/rapp-project/rapp-api/tree/master/python#rapp-platform-parameters)
@lpenteri ??

maciek-slon, 2016-07-28:
> That is due to the fact that the js API uses https as the default protocol and the RAPP Platform is launched at http

Yup, after changing to `http` tests are running (almost) properly.

JS call ([code](https://github.com/rapp-project/rapp-api/blob/devel/js/test/cloud_test.js#L155))
```
services.email_send( '', '', '', '', [], '', '', '' );
```

Web service output:
```
20 INVOKE.svc (email_send . #|JsObject [__proto__: JsObject] [extensible: #t] [properties: ()] [cmap: JsConstructMap] [elements: vector]|)
File "services/email_send/svc.js", line 75, character 28:
#    Fs.rmFile(req.files.file[0]);
#                            ^
TypeError: toObject: cannot convert "undefined"
    1. \@@callback \@@svcImpl hopscript, services/email_send/svc.js:75
    2. \@\@hopscript, src/rosbridge/src/core/bridge.js:203
    3. \@\@\@@ServiceController hopscript, src/rosbridge/src/core/ServiceController.js:107
    4. &thread3596
```

It's probably error in web service implementation when there are no files to attach - it tries to remove first file in [callback](https://github.com/rapp-project/rapp-platform/blob/master/rapp_web_services/services/email_send/svc.js#L75) and [onerror](https://github.com/rapp-project/rapp-platform/blob/master/rapp_web_services/services/email_send/svc.js#L83)

klpanagi, 2016-07-28:
This is indeed a bug in the web service implementation. Thank you for pointing this out @maciek-slon  :)

I am gonna prepare a master fix patch as soon as possible

lpenteri, 2016-08-18:
Sorry for taking so long to reply I was working on another project for some period and then was away on vacation. Indeed that is the reason behind the tests' failure. If you launch the platform configuring it to use https (I think you can do it changing the parameter [here](https://github.com/rapp-project/rapp-platform/blob/master/rapp_web_services/run.sh#L53)), they should work.
As for the reading of the platform's configuration parameters from a config file, I already have implemented it but I haven't found the time to commit it yet. I will dot so in the start of September along with the documentation of the API calls which is half finished. Although, I haven't included and I am not sure there is a reason to do so, a parameter for the protocol (http or https). Why would we even allow http communication when we have the safe https infrastructure already implemented?

closed 2016-10-28 by lpenteri