rapp-project/rapp-api #53: path_planning example test fails

dudekw, 2016-08-12:
Hi @alexge233, @klpanagi!

1) Path_planning example of rapp-api ends with error:
```
asio error: End of file
```   

full client log concerning `path_planning` service:
```
POST /hop/path_planning_plan_path_2d HTTP/1.1
Host: 155.207.19.229:9001
User-Agent: rapp_api/cpp/0.6.0
Connection: close
Accept: */*
Accept-Token: rapp_token
Content-Length: 476
Content-Type: multipart/form-data; boundary=bnEqpWAK5fONDmfE

--bnEqpWAK5fONDmfE
Content-Disposition: form-data; name="json"

{"map_name":"empty","robot_type":"NAO","algorithm":"dijkstra","":{"header":{"seq":"0","frameid":"frame_x","stamp":{"sec":"0","nsec":"10"}},"pose":{"position":{"x":"1","y":"1","z":"0"},"orientation":{"x":"1","y":"2","z":"2","w":"3"}}},"":{"header":{"seq":"1","frameid":"frame_x","stamp":{"sec":"0","nsec":"10"}},"pose":{"position":{"x":"2","y":"2","z":"0"},"orientation":{"x":"5","y":"6","z":"6","w":"7"}}}}

asio error: End of file
```

2) I found a bug in the request also. Client requested:
__start__ orientation: 1 2 3 4
__goal__ orientation 5 6 7 8

, but the log above shows: 

__start__ orientation: 1 2 2 3 
__goal__ orientation 5 6 6 7.

3) There is wrong upload_map HOP service name in [here](https://github.com/rapp-project/rapp-api/blob/cpp/cpp/includes/cloud/navigation/path_planning/path_planning.hpp). Following commit resolves this bug: ( https://github.com/rapp-project/rapp-api/commit/7a333d4ccbaff5865b5add58cd2c72d9ade050ea ).

alexge233, 2016-08-12:
@dudekw I haven't unit-tested the structs you've provided so be careful with them. Also I need to change the JSON library, so in version 0.7.0 towards which we're working on right now, will be very different from the current code, so again.

I've also added a fix to `struct quaternion` there was a small bug in the `treefy` method.

Regarding the third issue @klpanagi HOP/Platform should return `HTTP error 404` for wrong/non-existing URIs, instead of dropping/closing.

dudekw, 2016-08-17:
@alexge233 We really need stable c++ API implementation of the path planning service. Many following tasks depends on it (e.g. Hazard Detection Rapp tests, so we could finish Deliverable with @klpanagi ).

The path_planning include file needs following fixes:

```diff
diff --git a/cpp/includes/cloud/navigation/path_planning/path_planning.hpp b/cpp/includes/cloud/navigation/path_planning/path_planning.hpp
index e163a8d..3f14783 100644
--- a/cpp/includes/cloud/navigation/path_planning/path_planning.hpp
+++ b/cpp/includes/cloud/navigation/path_planning/path_planning.hpp
@@ -34,8 +34,8 @@ public:
         tree.put("robot_type", robot_type);
                tree.put("algorithm", algorithm);
 
-        tree.add_child("", start.treefy());
-        tree.add_child("", goal.treefy());
+        tree.add_child("start", start.treefy());
+        tree.add_child("goal", goal.treefy());
 
         std::stringstream ss;
         boost::property_tree::write_json(ss, tree, false);
@@ -44,7 +44,7 @@ public:
         post_  = "--" + boundary + "\r\n"
                + "Content-Disposition: form-data; name=\"json\"\r\n\r\n"
                           + ss.str();
-
+        post_ += "\r\n--" + boundary + "--";
                // set the HTTP header URI pramble and the Content-Type
         head_preamble_.uri = "POST /hop/path_planning_plan_path_2d HTTP/1.1\r\n";
                head_preamble_.content_type = "Content-Type: multipart/form-data; boundary=" + boundary;
@@ -127,6 +127,7 @@ public:
                     )
     : asio_http(), delegate_(callback)
     {
+               std::string boundary = rapp::misc::random_boundary();
                // multipart/form-data append JSON first
                post_  = "--" + boundary + "\r\n"
            + "Content-Disposition: form-data; name=\"map_name\"\r\n\r\n"
```
Unfortunately objects that are being sent to the cloud are type of string. The cloud requires some data fields  (e.g. robot poses, header.seq) to be float/int type. Could you fix it?

klpanagi, 2016-08-17:
@alexge233 , regarding https://github.com/rapp-project/rapp-api/issues/53#issuecomment-239406726, indeed the HOP Web Server should response with **HTTP 404** when requesting for non-existent Web Services (by uri) and it does not. Though this is a bug in the HOP Web Server implementation and should be reported [here](https://github.com/manuel-serrano/hop).

alexge233, 2016-08-18:
@dudekw current implementation can only use JSON to strings due to boost limitations. We're working on version 0.7.0 which will use a header only library (rapidjson) to do proper JSON parsing. Unfortunately for 0.6.1 I cannot change the JSON since it is the closest version to "stable".
I'll make the patch you've suggested, but bear in mind I have no way to test those classes because (a) we don't have a robot, and (b) I don't have the data you're using to test them.

maciek-slon, 2016-08-18:
@alexge233 Does it mean that it will not work since 0.7.0? You don't need the robot to test cloud services. You just have to call the service and see, whether it gives correct results (see the python tests for path planning services https://github.com/rapp-project/rapp-platform/blob/master/rapp_testing_tools/scripts/default_tests/path_planning_tests.py for examples).

alexge233, 2016-08-18:
@dudekw not with JSON PDT it won't work until 0.7.0, I've already hacked into removing string quotes, but some of the structs you've uploaded using nested boost. You're welcome to remove boost and implement it using another library, but our effort now is on 0.7.0 which should be ready early September.

alexge233, 2016-08-18:
@klpanagi Thanks, I've opened one [here](https://github.com/manuel-serrano/hop/issues/35)

klpanagi, 2016-08-18:
@alexge233  HOP actually responds with an **HTTP 404 Error**

Try this:

```bash
$ curl -I -X POST 155.207.19.229:9001/hop/mock
```

**Response headers**:

```http
HTTP/1.1 404 Not Found
Hop-Error: %22%01%1DService%20not%20found%20%22/hop/mock%22
Hop-Debug-Stack: (%01%02(%01%05%27%22%01%13%26pool-scheduler1055%3B(%01%02%27%22%01%04type%27%22%01%06server(%01%02%27%22%01%06format%22%01%03$%7Ea..
Cache-Control: no-cache
Pragma: no-cache
Transfer-Encoding: chunked
Connection: keep-alive
Content-type: text/html
Server: Hop
```

Also the python API is able to catch this HTTP response error succesfully ([here](https://github.com/rapp-project/rapp-api/blob/master/python/RappCloud/ServiceController/ServiceControllerBase.py#L198))

alexge233, 2016-08-18:
@klpanagi then the issue is with my handling of the response headers... Thanks.

maciek-slon, 2016-08-18:
I've hacked boost ptree and now it works with PDT. Now one have to explicitly state, that the value must be written with parentheses:

```c++
tree.put("frame_id", frameid_, s());
```

Also path planning and map uploading services has been fixed and now are working properly (maps are saved on the platform and returned path is parsed correctly). 

See commit e6d0587a59b19105a for changed files.

alexge233, 2016-08-18:
@maciek-slon you shouldn't have bothered, I'm in the processes of converting everything to use `rapidjson`. In any case, many thanks, use this for now so you can get your job done and I'll inform you on the transition to 0.7.0.

maciek-slon, 2016-08-18:
Early September is too late for us, so we've made it as it is, just like you said earlier this morning ;-)

> You're welcome to remove boost and implement it using another library

closed 2016-08-18 by alexge233