rapp-project/rapp-api #55: [cpp] Example data for `plan_path_2d`

alexge233, 2016-09-08:
Hi,

Can we please have a valid data example for the service call `plan_path_2d`?
We tried using arbitrary values and we didn't get an error but no response either.

@MariaRamos89 has committed d49882fa68ca3ea4151c29f5a5496a16e6693bb9 to branch `cpp_dev` , and tried using the [python example](https://github.com/rapp-project/rapp-api/tree/master/python/RappCloud/CloudMsgs#path-planning---plan-path-2d) which returns nothing.
The example that *did work* was the default values in the [path planning example](https://github.com/rapp-project/rapp-api/blob/cpp-wut/cpp/examples/path_planner.cpp) in branch `cpp-wut`.

1. If the coordinates are outside the boundaries, shouldn't we get an error? 
Preferably with the actual boundaries of the map?
2. If the pose is illegal, shouldn't we get an error?
3. How do you create those PNG map images, and do they have some kind of metric/distance embedded, or is the yaml descriptor that does that? The yaml does not seem to indicate much about the size of the map

Many thanks,
A.

etsardou, 2016-09-08:
I don't have a valid example but I'll try to answer the questions:
1,2. Yes, an error should be returned if no path can be produced. The reasons that may cause path planning to fail are 3: a) the robot pose to not be in unoccupied place (can be in occupied, unknown or out of bounds), b) the goal to not be in unoccupied place and c) no path can be created in unoccupied space. In all 3 situations you should get the corresponding error.
3) It is the yaml descriptor. The size of the map is computed from the size of the image, so yaml only contains how many meters each image pixel is [e.g.](https://github.com/rapp-project/rapp-platform/blob/master/rapp_testing_tools/test_data/path_planning/523_m_obstacle_2.yaml#L2)

alexge233, 2016-09-08:
@etsardou so regarding 3, the 0.02 resolution assigns 2cm per one pixel in that example?

Is there a RAPP or ROS manual on those algorithms and parameters, e.g., how do we generate a PNG for our office space and assign proper values?

Many thanks!

etsardou, 2016-09-08:
Yes, 0.02 is 2cm per pixel. In order to create a map you need a blueprint, either from an architectural design (where you have to estimate the resolution yourself) or by deploying a SLAM algorithm using a robot (where the map is created using a known resolution.

alexge233, 2016-09-08:
So we have the actual dimensions (we measured them by tape), and suppose we also measure the desk and table sizes, we set the resolution per pixel and align occupancies correctly?

I am also guessing a SLAM like [gmapping](http://wiki.ros.org/gmapping) would automatically generate the PNG? In the case of NAO we can't do that due to not having lazer sensors?

etsardou, 2016-09-08:
Yes, all correct, you need a robot with a laser sensor.

alexge233, 2016-09-08:
Great, many thanks, we'll write up some documentation on this.

dudekw, 2016-09-08:
@alexge233 @MariaRamos89  The service returns such information. The [planned_path](https://github.com/rapp-project/rapp-api/blob/cpp-wut/cpp/includes/objects/planned_path/planned_path.hpp) object consists following fields:

uint8_t `plan_found`;
std::string `planning_error`;
std::vector < rapp::object::pose_stamped >  `path`;

if plan_found == 1, than plan was found and you can find valid path in the `path` field. If your data is illegal, you will get error message in the  `planning_error` field.

3:
@etsardou is right. You can find example maps [here](https://github.com/rapp-project/rapp-platform/tree/master/rapp_path_planning/rapp_map_server/maps). 

We developed c++ example for the plan_path_2d service in the cpp-wut branch -> https://github.com/rapp-project/rapp-api/blob/cpp-wut/cpp/examples/path_planner.cpp

Yes, gmapping requires either laser or 3D camera like Kinect or Asus.

dudekw, 2016-09-08:
@alexge233 @MariaRamos89 [here](https://github.com/rapp-project/rapp-platform/wiki/RAPP-Path-planner#prepare-new-map-manually) is a short tutorial on `Prepare new map manually` .

alexge233, 2016-09-08:
@dudekw many thanks! I remembered you'd written something but couldn't find it.
Do you mind if we reuse that page for a tutorial?

dudekw, 2016-09-08:
@alexge233 feel free  reuse to any purposes. Let me know if you find something unclear.

closed 2016-09-17 by alexge233