rapp-project/rapp-api #58: After merge of the cpp_merge branch into cpp branch, the API doesn't build

lpenteri, 2017-01-31:
After applying the merge that I mention in the tittle, I tried to build the API along with the tests:
cmake .. -DBUILD_TESTS=ON
make

Make failed with the following output:
`
Scanning dependencies of target lrapp
[  9%] Building CXX object CMakeFiles/lrapp.dir/includes/config/config.cpp.o
[ 36%] Building CXX object CMakeFiles/lrapp.dir/includes/cloud/asio/asio_http/asio_http.cpp.o
[ 18%] Building CXX object CMakeFiles/lrapp.dir/includes/cloud/asio/asio_handler/asio_handler.cpp.o
[ 27%] Building CXX object CMakeFiles/lrapp.dir/includes/cloud/asio/asio_https/asio_https.cpp.o
[ 36%] Built target lrapp
Scanning dependencies of target rapp.0.6
[ 45%] Linking CXX shared library librapp.0.6.so
[ 45%] Built target rapp.0.6
Scanning dependencies of target serialise_test
Scanning dependencies of target service_test
Scanning dependencies of target object_test
[ 54%] Building CXX object CMakeFiles/service_test.dir/tests/service_test.cpp.o
[ 63%] Building CXX object CMakeFiles/object_test.dir/tests/object_test.cpp.o
[ 72%] Building CXX object CMakeFiles/serialise_test.dir/tests/serialise.cpp.o
In file included from /home/leizer/rapp-api/cpp/includes/objects.hpp:13:0,
                 from /home/leizer/rapp-api/cpp/tests/serialise.cpp:5:
/home/leizer/rapp-api/cpp/includes/objects/pose_stamped/pose_stamped.hpp:21:27: error: ‘msg_metadata’ in namespace ‘rapp::object’ does not name a type
       const rapp::object::msg_metadata header,
                           ^
/home/leizer/rapp-api/cpp/includes/objects/pose_stamped/pose_stamped.hpp:50:16: error: ‘msg_metadata’ in namespace ‘rapp::object’ does not name a type
  rapp::object::msg_metadata header;
                ^
/home/leizer/rapp-api/cpp/includes/objects/pose_stamped/pose_stamped.hpp: In constructor ‘rapp::object::pose_stamped::pose_stamped(int, rapp::object::pose)’:
/home/leizer/rapp-api/cpp/includes/objects/pose_stamped/pose_stamped.hpp:24:7: error: class ‘rapp::object::pose_stamped’ does not have any field named ‘header’
     : header(header), pose(pose)
       ^
/home/leizer/rapp-api/cpp/includes/objects/pose_stamped/pose_stamped.hpp: In member function ‘bool rapp::object::pose_stamped::operator==(const rapp::object::pose_stamped&) const’:
/home/leizer/rapp-api/cpp/includes/objects/pose_stamped/pose_stamped.hpp:36:17: error: ‘const struct rapp::object::pose_stamped’ has no member named ‘header’
   return (this->header == rhs.header) &&
                 ^
/home/leizer/rapp-api/cpp/includes/objects/pose_stamped/pose_stamped.hpp:36:31: error: ‘const struct rapp::object::pose_stamped’ has no member named ‘header’
   return (this->header == rhs.header) &&
                               ^
/home/leizer/rapp-api/cpp/includes/objects/pose_stamped/pose_stamped.hpp: In member function ‘boost::property_tree::ptree rapp::object::pose_stamped::treefy() const’:
/home/leizer/rapp-api/cpp/includes/objects/pose_stamped/pose_stamped.hpp:44:34: error: ‘header’ was not declared in this scope
         tree.add_child("header", header.treefy());
                                  ^
/home/leizer/rapp-api/cpp/tests/serialise.cpp: In function ‘int main(int, char**)’:
/home/leizer/rapp-api/cpp/tests/serialise.cpp:32:45: error: no matching function for call to ‘rapp::object::pose_stamped::pose_stamped(rapp::object::pose_metadata&, rapp::object::pose&)’
     rapp::object::pose_stamped ps(meta, pose);
                                             ^
In file included from /home/leizer/rapp-api/cpp/includes/objects.hpp:13:0,
                 from /home/leizer/rapp-api/cpp/tests/serialise.cpp:5:
/home/leizer/rapp-api/cpp/includes/objects/pose_stamped/pose_stamped.hpp:31:5: note: candidate: constexpr rapp::object::pose_stamped::pose_stamped(const rapp::object::pose_stamped&)
     pose_stamped(const rapp::object::pose_stamped &) = default;
     ^
/home/leizer/rapp-api/cpp/includes/objects/pose_stamped/pose_stamped.hpp:31:5: note:   candidate expects 1 argument, 2 provided
/home/leizer/rapp-api/cpp/includes/objects/pose_stamped/pose_stamped.hpp:28:5: note: candidate: constexpr rapp::object::pose_stamped::pose_stamped()
     pose_stamped() = default;
     ^
/home/leizer/rapp-api/cpp/includes/objects/pose_stamped/pose_stamped.hpp:28:5: note:   candidate expects 0 arguments, 2 provided
/home/leizer/rapp-api/cpp/includes/objects/pose_stamped/pose_stamped.hpp:20:2: note: candidate: rapp::object::pose_stamped::pose_stamped(int, rapp::object::pose)
  pose_stamped( 
  ^
/home/leizer/rapp-api/cpp/includes/objects/pose_stamped/pose_stamped.hpp:20:2: note:   no known conversion for argument 1 from ‘rapp::object::pose_metadata’ to ‘int’
[ 81%] Linking CXX executable object_test
[ 81%] Built target object_test
CMakeFiles/serialise_test.dir/build.make:62: recipe for target 'CMakeFiles/serialise_test.dir/tests/serialise.cpp.o' failed
make[2]: *** [CMakeFiles/serialise_test.dir/tests/serialise.cpp.o] Error 1
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/serialise_test.dir/all' failed
make[1]: *** [CMakeFiles/serialise_test.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 90%] Linking CXX executable service_test
[ 90%] Built target service_test
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
`

alexge233, 2017-01-31:
@etsardou WUT will have to update the names of the classes and objects. After that this should build.

maciek-slon, 2017-01-31:
Should be ok now

commit 20455c92 (2017-01-31): Fix #58
closed 2017-02-06 by alexge233